sinara-hw/Sayma_RTM #2: test: measure DAC-DAC phase drift

jbqubit, 2018-12-19:
Measure and document relative DAC phase noise ([ST3](https://github.com/sinara-hw/sinara/issues/601)) for pair of Sayma in same rack. Compare three outputs. 
- between channels on the same DAC
- between channels on adjacent DACs on the same RTM
- between channels on adjacent RTMs

Measure within 0.1 mHz of carrier using Microsemi 5125A.

jbqubit, 2018-12-19:
I asked analog Tech Support about drift. Here's their response. 

> What is the dependence of signal propagation on temperature? Suppose I have two AD9154 chips that have a common dac clock and are initially phase aligned and producing 400 MHz tone. What is the relative phase if one AD9154 temperature increases by 1 C? 

> I do not have a good answer here frankly. To quantify temperature drift, typically more than one part needs to be tested. We have little to no capacity to set this up at the moment. 
Two AD9154’s (or any other identical generics) would drift in the same direction, and the rate would be similar as well, assuming the same production lot. So the relative phase will likely stay the same, especially with only 1*C increase. The really interesting aspect would be if there is a temp gradient between two devices from two different lots (monte carlo of sorts).  
Temperature is inherently non-linear. So the slope due to a 1*C change will be different depending on the starting temp. Likely more so on the hot end. All in all, you would need to collect considerably more data than on a single part.  Going forward, we will have more temperature data on this class of parts.

So should check between several pairs of AD9154.

hartytp, 2018-12-19:
If you want to measure drift then the simplest way might be to just run the DACs at the same frequency then beat them against each other on a mixer and use a high-precision DMM to log the phase (given by the DC level). This is what we typically do and it works well.

closed 2019-05-27 by hartytp