sinara-hw/Sayma_RTM #3: test new PLL phase determinism

hartytp, 2018-12-21:


jbqubit, 2018-12-26:
@hartytp Where in the design review sequence does this need to take place? Do you have a proto board handy?

hartytp, 2018-12-26:
The results of this test will likely have no implications for the HW design because: (a) there are no other PLL families with decent noise that have the output divider inside the PLL FB loop, so AFAICT if this PLL family doesn't have deterministic phase then no PLL will (b) if we want to keep the HMC830 as a viable fallback option then we need to route the PLL output (after mux/fanout) to the RTM FPGA for phase detection (to allow us to reset the PLL until we get the correct startup phase), so the HW can still be made to work as desired even if the ADF PLL does not (it will just require more FW/FW development time).

That being said, the eval board is waiting for me in Ox and testing this is near the top of my priorities list atm.

closed 2019-02-27 by hartytp