sinara-hw/Sayma_RTM #4: bigger FPGA

hartytp, 2018-12-21:
Context:
- there is currently a 15T on Sayma RTM which is pretty tiny
- for DRTIO we need to add Satman. Currently, that should be fine but it will require some optimisation to reduce firmware size
- adding a bigger FPGA will make it easier to get DRTIO going and reduces risks
- we can always move back to a smaller pin-compatible FPGA in a future design revision to save costs

@sbourdeauducq what size of FPGA do you think we should go for? 50T? Since this is still a prototype, let's err on the side of having too big an FPGA rather than too small.

@gkasprow what's the biggest pin-compatible FPGA we can stick on Sayma RTM?

jordens, 2018-12-21:
The 50T is pin compatible. But as mentioned before that might not be needed at all since the bitstream is the same except for the chip name.

hartytp, 2018-12-21:
So, the suggestion is bitstream hacking to make a 50T bitstream work on the 15T (which is the same silicon)? What's the cost difference between the two FPGAs? Is it worth the hassle of bitstream hacking?

hartytp, 2018-12-21:
Anyway, right now my preference is to go with whatever is least effort and gets RF out of Sayma quickest. Once Sayma works and we've gauged the level of interest we can focus more on cost optimisation.

jordens, 2018-12-21:
Place a 50T for now if the availabliity is ok and keep in mind that moving back to a 15T will require software infrastructure changes.

sbourdeauducq, 2018-12-23:
Well, with some optimization it may even fit within the 15T limitations that Xilinx made up. But 50T on proto sounds good.

jbqubit, 2018-12-26:
50T sounds like the lowest risk option. @gkasprow @marmeladapk Is it too late for this change to the BOM?

gkasprow, 2018-12-28:
I will check if they are 100% compatible. Usually this involves generation of new symbol and comparison pin by pin because not always migration is that straightforward.

marmeladapk, 2018-12-28:
According to Xilinx they should are pin compatible, however we should rerun power estimator and adjust power delivery accordingly.

gkasprow, 2018-12-28:
They usually are, but It happened to me that some IOs were swapped during migration to bigger chip or not present in smaller packages. So it is not always that simple.

gkasprow, 2019-01-01:
Can we use XC7A35T-1CSG325 instead?

gkasprow, 2019-01-01:
It's hard to get XC7A50T in CSG325 package

sbourdeauducq, 2019-01-06:
Also use a faster speed grade, -1 does not meet timing for 150MHz RTIO frequency.

sbourdeauducq, 2019-01-06:
15, 35 and 50 are the same silicon so it doesn't matter so much which one you pick...

jbqubit, 2019-01-08:
@sbourdeauducq   Please concur if you're happy with XC7A35T-2CSG325. 

@gkasprow Are you comfortable that the risk is low enough for this change? 

- [ ] re-run power estimator 
- [ ] pin-by-pin check of new symbol

gkasprow, 2019-01-08:
The power supply is designed with a lot o margin. I can increase the current limit even by a factor of 5.

jbqubit, 2019-01-11:
@jordens Does your comment mean that @sbourdeauducq signed off on the use of 35T-3?

gkasprow, 2019-02-25:
XC7A35T-3CSG325 used.

kaolpr, 2019-07-18:
Long lead time for XC7A35T-3CSG325E (~14 weeks).

@sbourdeauducq @jordens Do you agree to use XC7A50T-3CSG325 only for the current production?

sbourdeauducq, 2019-07-18:
OK with me. The only important thing is the speed grade 3.

jbqubit, 2019-07-31:
XC7A50T-3CSG325 is fine by me.

jbqubit, 2019-08-19:
@gkasprow Please indicate how you're proceeding with this and close.

gkasprow, 2019-08-19:
We installed XC7A50T-3CSG325. It's the same silicon anyway.

closed 2019-08-19 by gkasprow