sinara-hw/Sayma_RTM #10: v2.0rc1 schematic cosmetics

hartytp, 2019-01-03:
Thanks for posting the new schematic revision @gkasprow! Posting small cosmetic issues here:
- [ ] remove annotation "Matching between the ... " and "SYSREF is matched to its CLK ... " 
- [ ] annotaiton "other diff lines: LVDS" is a bit odd, since all other diff lines are now explicitly named LVDS. Just remove this line from the annotation?
- [ ] clarify what "shared IOs" means. Shared with what?
- [ ] Missing overview page?

hartytp, 2019-01-03:
- [ ] add annotation with frequency of OSC2

hartytp, 2019-01-03:
- [ ] HMC_RESET -> HMC830_RESET

hartytp, 2019-01-03:
- [ ] remove "SYSREF is matched to its CLK (a.k.a. DEVICECLK) to better than 100 ps" this no longer makes sense

atcher0, 2019-01-14:
- [ ] DAC_CFG buses are not zero-indexed
- [ ] HMC_RESET and HMC_SPI_GPIO lines are not used (CLK_DIST_CFG bus)
- [ ] HMC830_SPI_SEN != HMC_SPI_SEN (clock distribution page)
- [ ] no information about FPGA configuration settings
- [ ] many FPGA not connected pins are not marked NC
- [ ] OSC2 is placed on schematic without part number (+ no language describing use case) 
- [ ] pages are not numbered consecutively

marmeladapk, 2019-01-14:
- [ ] Add annotations about TPSM84824: switching frequency (450 kHz) and soft start (2,4 ms)

gkasprow, 2019-02-26:
@hartytp what overview page would you like?

hartytp, 2019-02-26:
Something with a system BD that makes it clear what blocks we have in the design and how they connect together. Whatever you think is best though.

hartytp, 2019-03-16:
@gkasprow the new schematics are really beautiful. Very nicely annotated (comments about clocking regions for LVDS lines, colour coding for busses etc). Really well done on that :)

hartytp, 2019-03-16:
We can delete Weida's diagram from the schematics, since there isn't anything in there that isn't already in the schematics.

hartytp, 2019-03-16:
- [ ] worth adding an annotation on the HMC7043 that the FETs are used to avoid sending broadband noise to the FPGA before the IC is configured.

gkasprow, 2019-03-16:
Thanks. It's a side effect of switching to Altium :)

hartytp, 2019-03-16:
- [ ]  maybe add an annotation about the purpose of the bleed FETs on the HMC 830 supply (took me a minute to remember why they're there)

marmeladapk, 2019-04-05:
- [ ] P5V0_LDO_IN  (near IC15) should be P4V0...

hartytp, 2019-05-27:
@gkasprow are you happy with the overview page here? If so, let's close this issue.

marmeladapk, 2019-08-30:
[ ] Rename xVDD12 to xVDD13 on dac schematics?