sinara-hw/Sayma_RTM #13: Sysref driving

hartytp, 2019-01-03:
@gkasprow how do you want to drive the AD9154 SYSREF? IIRC it should ideally be driven DC coupled with a common-mode voltage of 600mV (but someone should check this during the design review). That's not directly compatible with LVPECL without a RC termination network.

hartytp, 2019-01-03:
- [ ] even though this will be DC coupled, please ensure that there are suitable pads that can be used to add AC coupling capacitors later on if we find it necessary

hartytp, 2019-01-09:
cf https://github.com/sinara-hw/Sayma_RTM/issues/17

hartytp, 2019-01-09:
NB fwiw, the eval board has the SYSREF AC coupled, with a test point used to drive/probe the DC level.

IIRC the data sheet implies that SYSREF should be DC coupled where possible.

I would like to ensure that we can drive SYSREF either DC or AC coupled (choice by population options), with DC coupling being the default choice.

gkasprow, 2019-01-09:
If SYSREF is going to be a non-50% duty cycle, we must use DC coupling. We must make sure DAC accepts LVPECL common model

hartytp, 2019-01-09:
> We must make sure DAC accepts LVPECL common model

I don't think it does. IIRC the pin self-biases to 600mV (there is a figure in the data sheet that shows the simplified schematic for the input configuration), which is the common-mode voltage it's designed to work at. I don't think the common-mode voltage of LVPECL/CML will do the job.

marmeladapk, 2019-01-14:
Now SYSREF is AC coupled (and capacitors don't have values on schematics).

gkasprow, 2019-02-27:
well, not really. The high level of LVPECL can be even 2.4V. So out of the common mode of the DAC.
Do AC-coupling cause any issue? Certainly, we are not over-driving the inputs. I added the series 50Ohm resistors to lower the clock amplitude.

hartytp, 2019-02-27:
@gkasprow the thing I'm conscious of is that the default (self-biasing/AC coupling) DAC input common-mode bias voltage is only 600mV. I believe that it can work with other common-mode voltages, as indicated by the data sheet, but requires a larger swing to operate correctly. There is an equivalent circuit of the input stage that shows how this works.

So, I'm wary of DC coupling LVPECL in to the DAC unless we really think about it.

gkasprow, 2019-02-27:
So let's leave it as it is now.

hartytp, 2019-02-27:
@gkasprow can you confirm:
1. SYSREF is DC coupled into the DAC (although, let's make sure there are pads for AC coupling). IIRC currently SYSREF is AC coupled
2. DAC clocks and SYSREF are correctly terminated
3. which HMC7043 output standards is this designed to use (let's make sure that's annotated on the schematic)
4. that the common-mode voltage and voltage swing are appropriate for both the DAC clock and SYSREF

Once that's done we can close this issue

gkasprow, 2019-02-27:
to make SYSREF DC-coupled I have to lower the LVPECL output to less than 2V.
LVDS mode does not meet minimum 400mVpp DAC requirement.
So either we use CML mode that has VCC-1.6V common mode voltage or LVPECL with resistive divider.
I suggest using direct CML mode with series resistors that will reduce the swing
[image]
two 50R resistors should do the job and preserve common mode

gkasprow, 2019-02-27:
Let's keep the 200R DNP so we can always use them and replace 49R9 with 100nF and revert to AC coupling.

gkasprow, 2019-02-27:
[image]

closed 2019-02-27 by gkasprow

hartytp, 2019-02-27:
Do you need to add termination to Vcc at the DAC then?

hartytp, 2019-02-27:
(if you're using CML for SYSREF that is)

hartytp, 2019-02-27:
Also, are you sure that 3.3-1.6=1.7V common-mode will work for the DAC SYSREF?

gkasprow, 2019-02-27:
It accepts it up to 2V so with reduced swing should work.

gkasprow, 2019-02-27:
HMC has internal termination for CML mode
[image]

hartytp, 2019-02-27:
Okay. Can you leave resistor pads so that we can make a divider between Vcc and GND if we need to modify the voltage swing/DC bias point.

gkasprow, 2019-02-27:
OK, I will add 2 DNP resistors to GND to make such divider.