sinara-hw/Sayma_RTM #17: delay line termination

hartytp, 2019-01-03:
I don't understand the termination for the digital delay line:
- Why are R127 and R128 used for *DC* coupled LVPECL?
- Why is R129 necessary?
- Don't we need to apply a Vcc-2V=1V3 DC bias to the Vtt pin? Just use one of the standard RC networks for this

hartytp, 2019-01-03:
Similarly for the DFF:
- IIRC the CK and D lines of the MC100EP31DTG are **not** internally 50R terminated (they just have 75k) pull downs. So external termination is required
- there is a version of the DFF that has a differential LVPECL clock input. We might as well use that since we have a differential driver available (don't worry about this if it's hard to do, but it would be nice since better emi resistance)
- The CK input needs appropriate DC biasing since it's DC coupled (this could be AC coupled if it's easier)

gkasprow, 2019-02-25:
not relevant anymore

closed 2019-02-25 by gkasprow