sinara-hw/Sayma_RTM #21: adf PLL review

hartytp, 2019-01-03:
Will complete this after testing. Howerver....
- [ ] remove the bias inductors. They increase the power by 1dB at high frequencies but kill the power at low frequencies. Not worth the hassle
- [ ] review the HMC1060LP3E connections
- [ ] @gkasprow if possible, I'd like to scrap IC45. The ADF PLL has two outputs, which we can use to drive the DACs directly
- [ ]  after testing the ADF pll we should consider scrapping the HMC830. NB if we keep it then we need to route the HMC830 output to the RTM FPGA for phase measurements to synchronise the PLL output divider

hartytp, 2019-01-03:
If we scrap the mux then just use some weak capacitors to pick off some power for diagnostic UFLs (same as you're doing for the HMC830)

jbqubit, 2019-01-08:
> after testing the ADF pll we should consider scrapping the HMC830

No. The project plan calls for layout of HMC830. Wait until after v2.0 to jettison HMC830.

hartytp, 2019-01-08:
Let's see how I get on with the AF PLL before making a final call on this. The clock path is critical and the more components we cut out the more likely it is to actually work.

hartytp, 2019-01-08:
Also, as Greg's pointed out elsewhere, while it's easy to add things to the schematic, doing a good layout with minimal cross-talk is much harder and more time consuming (not to mention, quicker to review and test).

Right now, my preferred option would be this:
- implement the ADF as the only PLL, connecting the two outputs to the two DACs directly (subject to testing by me, verifying logic levels etc)
- use solder jumpers to allow the DAC clocks to be driven by an external source
- add a gpio header with power that connects to the FPGA
- if we do have any issues then we can bodge on a simple PCB with just an HMC830 + output buffer that connects via header/coax to the board. That gives us a simple risk-mitigation plan without having to stuff the main Sayma design with layers of cruft.

gkasprow, 2019-01-09:
We have a lot of space on board and HMC part is already routed.

hartytp, 2019-01-09:
@gkasprow Okay. I was just hoping to scrap the clock mux.

How about a different compromise:
- We connect the two ADF outputs directly to the two DACs (no fanout buffer)
- we use solder jumpers to allow the ADF outputs to be disconnected and the DAC clocks to be driven from UFL connectors (useful anyway for injecting external DAC clocks and other diagnostics)
- we leave the HMC830 and fanout buffer
- we connect the fanout buffer outputs to UFL connectors
- that way, to use the HMC830 users just add some short UFL jumpers. This is inconvenient, but gives a good redundancy plan

hartytp, 2019-01-09:
Note that the current design is somewhat problematic for DAC clock frequencies below 1.5GHz when using the HMC830 since the output divider is not synchronised.

It should be possible to get around this by using the AD9154 as a phase detector to infer the startup phase of the HMC830 output dividers and reset the PLL until the dividers start with the right phase, but that has not been demonstrated yet. Without testing, it is not clear whether this will work, so it's a serious risk with the HMC830.

We also have not demonstrated the HMC830 operating reliably yet. To get it to work reliably would require extra power-cycling FETs, which aren't in the current RTM schematic IIRC. Even once they're added, we won't know for sure that it all works without further testing.

All in all, it's not clear to me that the HMC830 offers a lower risk path, since we have not yet demonstrated it working really robustly. Moreover, it has known issues (unsynchronised output dividers) so I feel we're better focusing on getting the ADF PLL working really well and only keeping the HMC830 as a real last resort.

gkasprow, 2019-02-25:
it looks that this issue is no more relevant and we can close it.

closed 2019-02-25 by gkasprow