sinara-hw/Sayma_RTM #22: HMC830

hartytp, 2019-01-03:
verify that connections match the eval board schematic

marmeladapk, 2019-01-11:
I only wonder how will HMC830 behave, when it gets high state on SPI lines while powered down. Clock could also be problematic.

marmeladapk, 2019-01-14:
Credit @atcher0, this one is pretty critical, ping @gkasprow 

> HMC830_SPI_SEN != HMC_SPI_SEN (clock distribution page)

gkasprow, 2019-02-25:
One must simply make sure that when EN_PWR_HMC830 is low, other SPI signals must be low as well.
I added discharge resistors to the HMC830 supply rails and transistor that disables ADCLK948 pulldowns.
So we make sure that the HMC supply will fall to zero.

[image]
[image]

closed 2019-02-25 by gkasprow