sinara-hw/Sayma_RTM #25: FPGA pinout

jordens, 2019-01-11:
@gkasprow for the stub ARTIQ port we need the FPGA pinout

gkasprow, 2019-01-13:
Here you go.
[Sayma_RTM_pins.txt](https://github.com/sinara-hw/Sayma_RTM/files/2753098/Sayma_RTM_pins.txt)

gkasprow, 2019-02-25:
here is an updated version. Some missing labels added.
[FPGA_pins.xlsx](https://github.com/sinara-hw/Sayma_RTM/files/2902982/FPGA_pins.xlsx)

closed 2019-02-25 by gkasprow

sbourdeauducq, 2019-02-26:
@gkasprow Why is it XLSX now? Is that because of the Altium conversion?

sbourdeauducq, 2019-02-26:
And what is the exact PDF schematics that corresponds to this file?

gkasprow, 2019-02-26:
this is extracted from Altium tool. I can extract the netlist as well if U like.

gkasprow, 2019-02-26:
I can also generate csv file

gkasprow, 2019-02-26:
@sbourdeauducq The schematic will be in releases in a moment.

sbourdeauducq, 2019-02-27:
@gkasprow Do you confirm that those schematics:
[image]
correspond ***exactly*** to those files:
[image]

gkasprow, 2019-02-27:
In a few hours I will publish both pdf and xls of new revision with several
issues sorted out.
Greg