sinara-hw/Sayma_RTM #30: DAC clock PLL

hartytp, 2019-01-23:
To do:
- [ ] all sign off on choice of HMC830
- [ ] either add power cycler or use LDO with enable pin and add drain resistors
- [ ] add DNPd pads for 3rd order active loop filter
- [ ] route output of DAC clock fanout to FPGA via DFF. DFF clock driven by DAC clock, data input driven by FPGA to allow gating. DFF output driven to diff input in FPGA. Used for S/H validation. @sbourdeauducq to sign off on this part of the design, check frequency range, voltage standards on FPGA
- [ ] add mounting holes and IDC that can be used to for external clock board. Add UFL/MMCX to unused mux input

hartytp, 2019-01-29:
@gkasprow are you happy with all the items on the list above?

One thing we didn't discuss on the phone was the HMC830 loop filter. Based on our experience here, passive loop filters are generally not the best for phase critical applications, so it's worth leaving ourselves the option of adding an active loop filter. Can be DNPd pads for now and we'll prototype it once we have the new hw.

hartytp, 2019-02-26:
>     * route output of DAC clock fanout to FPGA via DFF. DFF clock driven by DAC clock, data input driven by FPGA to allow gating. DFF output driven to diff input in FPGA. Used for S/H validation. @sbourdeauducq to sign off on this part of the design, check frequency range, voltage standards on FPGA

@sbourdeauducq I assume we no longer need this, as the DAC seems to behave fine at max clock rate...

hartytp, 2019-02-26:
> add mounting holes and IDC that can be used to for external clock board. Add UFL/MMCX to unused mux input

If we're happy with the HMC830, we can also scrap this item -- although, I still think that exposing some GPIO and supply voltages on pin header would be a very good idea

gkasprow, 2019-02-27:
we will keep 2V5 IO pin header so we can use it for any purpose.

gkasprow, 2019-02-27:
What loop filter would you like for HMC? Active one? So shall I add RC place holders and opamp?
Any suggestions on opamp?

hartytp, 2019-02-27:
I'd like to have a passive loop filter (same one we currently have) fitted by default, but the pads there so we can add an active loop filter later on if testing indicates that it would help. Just use a generic SOIC8 OpAmp footprint for now.

@WeiDaZhang is thinking about active loop filters right now, and will post a recommendation for a topology to use soon.

closed 2019-03-05 by hartytp