sinara-hw/Sayma_RTM #31: Sayma RTMv2 mechanics

gkasprow, 2019-02-23:
I'm doing component placement of the v2 board. Unfortunately, I had to almost entirely unroute the RTM board - only the supply section and DACs remained intact. 
So, this is an occasion to optimize the component placement.

- I will place part of the clocking section mostly on the top layer. Only HMC7043 would be placed on bottom side as it was originally. I want to avoid routing long traces for the DAC clock.
- The clock input SMA must be in the middle of the front panel. There is no other way to install it.
- one would need to remove the front panel to replace the AFE modules. Another option is to unscrew the panel, remove the module keeping the panel attached to the RTM board.
- I will move the FPGA towards board centre to make routing LVDS lines easier
[image]

gkasprow, 2019-02-23:
I can place the SMA input connector in line with other SMAs using the right-angle connector.

gkasprow, 2019-02-23:
Actually, it's hard to make it ideally in-line even with low profile RA SMA connector
[image]

hartytp, 2019-02-23:
I’m not fussed about the clock SMA being inline with the other SMAs. Can even be MCX if that helps. Let’s just make sure all clock traces are as short as possible and we keep plls away from noisy components like SMPSs

closed 2019-02-25 by gkasprow