sinara-hw/Sayma_RTM #35: more SYSREFs to FPGA

sbourdeauducq, 2019-02-27:
Right now there is only one SYSREF (and it seems to be not even properly connected).
Ideally, we want two SYSREFs on each FPGA (RTM and AMC). See here:
https://github.com/sinara-hw/sinara/issues/574#issuecomment-467014358

gkasprow, 2019-02-27:
I can scrap one GTH clock that goes to the 227 bank.

sbourdeauducq, 2019-02-27:
Commented in the wrong issue?
This is about the SYSREF signals on the RTM. I believe this is resolved.

sbourdeauducq, 2019-02-27:
Or you mean, removing a GTH clock to make room for a SYSREF in the AMC/RTM connector?

gkasprow, 2019-02-27:
Yes. I connected 2 SYSREF signals to the RTM FPGA and only one to AMC FPGA.

sbourdeauducq, 2019-02-27:
OK in that case yes:
* remove GTH clock to bank 227
* replace with SYSREF that goes into the same bank as the CDR clock (We will try to use DDMTD this time, not simple 125/150 sampling).

sbourdeauducq, 2019-02-27:
And make sure that the CDR clock goes to a GC.

gkasprow, 2019-02-27:
OK, the LVDS30 pair is used instead of GT clock
[image]

it is routed to the HMC output from one side
[image]

and AMC FPGA input from other side, just about the CDR clock input
[image]

closed 2019-02-27 by gkasprow