sinara-hw/Sayma_RTM #36: bank 34 does not have voltage indication on schematics

sbourdeauducq, 2019-02-27:
[image]
[image]
[image]

sbourdeauducq, 2019-02-27:
Also there is something not right with that voltage. The oscillator is 3.3V, the HW_ID resistors are 2.5V and LVDS requires 2.5V (we have only HR banks with LVDS_25). I suppose the oscillator should output 2.5V levels instead.

sbourdeauducq, 2019-02-27:
Also, is the resistive divider that brings the 50MHz oscillator output to 2.2V something that you have done before and which yields good SI?
What about using a LVDS oscillator instead?

gkasprow, 2019-02-27:
fixed. It is 3.3V

sbourdeauducq, 2019-02-27:
3.3V doesn't work for LVDS.

sbourdeauducq, 2019-02-27:
Did you mean 2.5V?

gkasprow, 2019-02-27:
Bank 27 is connected to 3.3V to provide signaling for HMC chips and mezzanines. Other banks are connected to 2.5V and LVDS are located there.

closed 2019-02-27 by gkasprow