sinara-hw/Sayma_RTM #37: resistive divider for 50MHz oscillator

sbourdeauducq, 2019-02-27:
@gkasprow is the resistive divider that brings the 50MHz oscillator (OSC1) output to 2.2V something that you have done before and which yields good SI?
What about using a LVDS oscillator instead?

sbourdeauducq, 2019-02-27:
Alternatively, scrap that oscillator - we can use the path through the GTP (OSC2) like we do on Kasli.

gkasprow, 2019-02-27:
it is close to the FPGA. Impedance matching could be better. I can increase the R37 and keep effective R of roughly 50hm. CMOS oscillators are much cheaper than LVDS ones. I found it useful for testing.

gkasprow, 2019-02-27:
this gives Z=51Ohm
[image]

closed 2019-02-27 by gkasprow

jordens, 2019-03-03:
I'd actually scrap OSC1.