sinara-hw/Sayma_RTM #43: observing DAC output for debugging

sbourdeauducq, 2019-02-27:
What is the plan for looking at the DAC outputs during initial board bring-up?
Should we make minimalistic AFEs with just balun transformers and connectors?

gkasprow, 2019-02-27:
There is already one - the TestMod.

gkasprow, 2019-02-27:
Look [here](https://github.com/sinara-hw/TestMod/wiki)

sbourdeauducq, 2019-02-27:
Okay, good. We will need them for firmware development.

closed 2019-02-27 by sbourdeauducq

gkasprow, 2019-02-27:
I have 4 pcs so far.

sbourdeauducq, 2019-02-27:
We need at least 2 for DAC synchronization tests. At least 3 would be more convenient.

hartytp, 2019-02-27:
> I have 4 pcs so far.

Is that with the proper stack up? If so, can you lend me one at some point to stick it on the VNA?

gkasprow, 2019-02-27:
I didn't measure them yet so I cannot confirm. They are supposed to be with the right stackup.

hartytp, 2019-02-27:
Cool. Well, if you send me one I'm happy to have a look.

gkasprow, 2019-02-27:
OK.