sinara-hw/Sayma_RTM #45: spare GTP channels

sbourdeauducq, 2019-02-27:
https://freenode.irclog.whitequark.org/m-labs/2019-02-27#1551260565-1551262342;

gkasprow, 2019-02-27:
Interesting idea.
 But we decided that we will utilize all RTM connector pins to LVDS for AFE.
Is there any use case for RTM FPGA having access to the AMC side FMC?

jordens, 2019-02-27:
I meant routing the GTPs to the RTM-AFEs (via their FMC connector), not to the AMC-FMC.

gkasprow, 2019-02-27:
OK, this makes sense. We can do that.
Just don't use FMC name. It stands for the FPGA Mezzanine Card. I call these AFE connectors :)

gkasprow, 2019-02-27:
which SATA shall I keep?

gkasprow, 2019-02-27:
we can use pins e31/e32 and b31/b32

gkasprow, 2019-02-27:
I mean master or slave SATA?

jordens, 2019-02-27:
Master SATA (the one that connects with a crossover cable to J16 on the AMC and to Kasli) is fine. There are crossover cables IIRC easily available.

gkasprow, 2019-02-27:
done

closed 2019-02-27 by gkasprow

jordens, 2019-03-03:
And in general we should remove those "slave SATA" things everywhere and turn them all into a single SATA pinout. The RX-TX crossover is easily done in the cable and does not restrict the connectivity options.

elektron314, 2020-07-07:
does sata cable really have crossover inside of it?

sbourdeauducq, 2020-07-07:
You can find some special cables that do; they are rare but they exist.

elektron314, 2020-07-07:
so if I plan to connect two devices with SATA I need to make TX at 2,3 pins and RX at 5,6 pins on one side, and TX at 5,6 and RX at 2,3 on the other side?

sbourdeauducq, 2020-07-07:
I would recommend buying a commercial crossover cable, SATA cables and connectors are messy to tinker with.

elektron314, 2020-07-07:
yes, exactly commercial. But I want to make standard connection, not rare. So do I need to make inverted connection on pcb side because almost all cables are straight?

sbourdeauducq, 2020-07-07:
IIRC I ordered them from https://www.techcable.com/sas-mini-sas-cables/sata-cables/

sbourdeauducq, 2020-07-07:
Yes, almost all cables are straight.

elektron314, 2020-07-07:
thank you, mister! you saved our company!