sinara-hw/Sayma_RTM #48: drive AFE REF clock inputs from HMC7043 and not from upstream ADCLK948

jordens, 2019-03-03:
Just a choice to consider:

* there are two unused HMC7043 outputs
* would allow a bit of power saving and configurability
* would allow configurable REF frequency for the AFE
* might be slightly better or slightly worse phase noise (si549 vs hmc830 and external ref vs hmc830)

hartytp, 2019-03-05:
hmmm...The use case I have for these ref lines would be supplying the reference for a 3GHz LO PLL on the AFE mezzanine. The long-term phase stability of the LO is crucial (more important than the DAC clock due to higher frequency), so I'd like to keep the number of components in the clock path to an absolute minimum. Adding the HMC830 + HMC7043 essentially act as a fanout for the 125MHz seems like a lot of extra components that could add extra drift. It may be fine in practice, but that's not obvious without testing.

Obviously, if other people need the extra flexibility of the HMC830 + HMC743 then that could outweigh these risks, but unless anyone presents a concrete use-case, I'd prefer to just use a fanout buffer.

gkasprow, 2019-03-05:
If someone wants a high-frequency clock on the mezzanine, it can always install there a copy of HMC chipset when needed. I can also add an assembly option but there will always be some crosstalk...

jordens, 2019-03-06:
Would the upstream recovered clock stability be degraded by this? It's going to be limited by the distribution and DDMTD recovery stability.

hartytp, 2019-03-06:
> If someone wants a high-frequency clock on the mezzanine, it can always install there a copy of HMC chipset when needed. I can also add an assembly option but there will always be some crosstalk...

Yes, for MixMod (which is currently the only place we have plans to use this reference on the AFE) we will install an HMC830.

> Would the upstream recovered clock stability be degraded by this? It's going to be limited by the distribution and DDMTD recovery stability.

Until we've done a thorough audit of the stability/noise contributions of each component in the clock chain, it's not clear what the limitations will be.

nonetheless, keeping clock chains as simple as possible is generally good practice. So, the question is whether we envisage any scenarios where the HMC7043 would be helpful. Right now, I don't see any. The only AFE currently intended to use the reference is MixMod, which will have an HMC830 on it to produce the LO (the LO will be at a higher frequency than the DAC clock, so we can't produce it by directly from the HMC7043 output).

Let's try to just keep this as simple as possible...

jordens, 2019-03-06:
I'd operate under the assumption that DDMTD will worse than the HMC830 on those timescales. I don't see how using the upstream ports instead of the downstream ports is simpler. Unloading and maybe even eliminating that fanout would definitely be a simplification. And MixMod is quite a bit down the road. The simplest possible and best stability approach would be to feed MixMod externally. But well. IMO not worth arguing now.

closed 2019-03-06 by jordens

hartytp, 2019-03-06:
> And MixMod is quite a bit down the road

Right now any application at all for this reference pin is quite a bit down the road. BaseMod certainly won't use it.

Let's just stick with the fanout. Once we have some data from Sayma v2.0 and a clearer picture of potential use-cases we can revisit this decision.