sinara-hw/Sayma_RTM #49: HMC830 active loop filter

hartytp, 2019-03-05:
Courtesy of @WeiDaZhang, let's use this loop filter topology:

[image]

Comments:
- Let's stick with the current 3rd order loop filter for the next production run, but add pads for the active 6th order filter. These pads allow us to use either a passive 4th order loop filter or an active 6th order loop filter, see component values below. Please add component choice table to the schematic as an annotation
- NB the HMC830 PFD noise dominates the in-loop noise for the final system assuming a good reference source like a Wenzel oscillator or SI549. So, the optimal loop filter does not depend on the reference source
- The OpAmp could be a LT6018. Ideally, let's run this from a 5V supply
- NB since the charge pump output has a high impedance, we're actually not very sensitive to noise on the OpAmp non-inverting pin, so we don't need a low noise reference to generate this supply; RC filtering from a low-noise LDO is fine
- The higher-order filters used here give very good rejection of out-of-loop spurs (e.g. from WR)

 | Passive 4th Order | Active 6th Order
---|---|---
R0 | 1.1k | 0R
C1 | 3n | 51p
R2 | 5.6k | 1.1k
C3 | Short | 2n
L4 | DNP | 8.2u
Rp | 1.1k | 1.1k
C2 | 4.3p | 2.7n
Cp | 30p | 24p
C5 | 1.3p | 3.3n
R3 | Short | 51
Cb1,2 | DNP | 0.1u//100p
Rb1,2 | DNP | 1k
R4 | 22k | DNP

hartytp, 2019-03-05:
To remind, the main motivation for using the active loop filter will be if we find phase stability in the passive loop filter. So, the methodology will be to test using the current loop filter and not bother with the active loop filter unless we find a reason to.

@gkasprow please make sure that the loop filter is as compact as possible (small loop area) and routed away from all noise sources (SMPSs etc) to minimise pickup. Thanks!

hartytp, 2019-03-05:
Simulation files for these loop filters...

[Archive.zip](https://github.com/sinara-hw/Sayma_RTM/files/2932831/Archive.zip)

gkasprow, 2019-03-05:
@hartytp are U sure we want inverting amplifier filter?

WeiDaZhang, 2019-03-06:
> are U sure we want inverting amplifier filter?

Very good point. Thanks @gkasprow 
All the active filter topologies provided in the Hittite PLL software are inverting.
And, these loop filters can be simulated with HMC830.
So, I was assuming the HMC830 can work with inverting active filter.

Yes, and no, after checked.
It doesn't work by default, a bit in the registers needs to be set.
[image]
But overall, it should be fine.

hartytp, 2019-03-06:
Right, the HMC830 can nominally work with active loop filters by setting the pd_phase_sel bit, so I do not anticipate a problem here. But, I haven't tested this in action yet. Will test once the new hw arrives.

gkasprow, 2019-03-11:
@hartytp don't you mind if I replace LT6018 with something that has standard pinout, like ADA4528?
I also have LT6018 in my libs so it doesn't make any difference which I use but later on, it would be hard to replace it with another type.
The loop filter schematic is here. 0R resistors (R148, R142) will not be mounted.
[image]

hartytp, 2019-03-11:
@weidazhang ^

hartytp, 2019-03-11:
Afaict it’s fibe. Need to check the common mode input range and noise

gkasprow, 2019-03-11:
ADA4528 has much higher noise (5.6 nV/√Hz) but it is rail to rail.

gkasprow, 2019-03-11:
LT6018 needs at least 8V supply,  input range is from 3V above the supply rail, so we would need a negative supply with possible power sequencing issues. I'd prefer opamp supplied from 5V

hartytp, 2019-03-11:
If we're happy going for a TSOT package, there are a few nice options like LT623{036}

Anyway, I'll leave the final call to @WeiDaZhang since he's thought about this more.

gkasprow, 2019-03-11:
SO-8 is also fine.

WeiDaZhang, 2019-03-12:
No objection to ADA4528 nor LT6236, though ADA4528 is more compatible with other OpAmps. the OpAmp noise is well covered by the PD noise.

closed 2019-04-08 by gkasprow