sinara-hw/Sayma_RTM #50: DAC output -> fast comparator -> RTM FPGA input

sbourdeauducq, 2019-03-09:
Is it realistic to add such a path on Sayma RTM?

* Useful as a built-in self-test to check that DAC outputs are toggling *and* synchronized to the FPGA.
* One path for each DAC.
* AMC FPGA generates test pattern on JESD204 which is timed by DDMTD core on RTM FPGA.
* RF switch on AFE can be used to disable output during BIST.
* The fast comparator could potentially be a hacked LVDS input buffer in the FPGA, with proper signal conditioning.
* Maximum DAC frequency is 1.2GHz. Maximum data rate received correctly by the FPGA is rated at 1250MBps (625MHz). We can either add a RF switch to avoid overloading the FPGA input, or just hope that it won't cause any ill effects. Based on past experience with Sayma, I prefer the former option.
* If we use an external comparator, the RF switch action can be effected by driving the other comparator input out of DAC range.

hartytp, 2019-03-09:
Comments/thoughts:
- What if we put this extra DFF on BaseMod/TestMod rather than on Sayma RTM directly? That way we can easily make modifications later on, or remove it once we're satisfied that it's no longer needed.
- I assume that what we'd do is use resistors to pick off a fixed amount of power from one channel on each DAC. Route that to a fast comparator on the AFE mezzanine. Ideally use a comparator with enable input. Can we get away with AC coupling? Then route the comparator output through the FMC connector to a LVDS input on the RTM FPGA.

hartytp, 2019-03-09:
Both OnSemi and ADI have ranges of fast comparators. The amount of power we pick off can be really quite small if done properly, so this needn't interfere with the DAC's normal operation.

Anyway, if we are happy putting this on the AFE then we don't need to delay freezing the RTM schematics, and it's also lower risk since it's cheap to make new AFEs if we mess it up...

sbourdeauducq, 2019-03-09:
Yep, putting it on the AFE sounds good.
If we put an I2C EEPROM on the AFE to identify it, we can even autodetect whether those comparators are there, and skip the test when they aren't.