sinara-hw/Sayma_RTM #57: Sayma v2: more test points

jbqubit, 2019-03-13:
From #408. @gkasprow More test points on Sayma v2.0, please! Particularly on all the SPI buses.

hartytp, 2019-03-14:
Could you have a go at this today please? I’ll do my review after it’s done

jbqubit, 2019-03-15:
@gkasprow Please implement change or give update. We'd all like to see pending changes to RTM complete and design frozen.

gkasprow, 2019-03-16:
Let's add missing test points to the issues and go on with the review.

hartytp, 2019-03-16:
Most important are the HMC830 and HMC7043 SPI. I'd also ideally like to have 0R resistors between the HMC chips and the FPGA for debugging (e.g. if we need to disconnect some signal).

Most analog power lines should have test points. 

All I2C signals

hartytp, 2019-03-22:
AFE power supplies, I2C

gkasprow, 2019-04-08:
fixed

closed 2019-04-08 by gkasprow