sinara-hw/Sayma_RTM #60: LDO Vset bypass

hartytp, 2019-03-16:
All the LDO Iset capacitors are electrolytic. What's the reason for that when we've always used ceramic for this elsewhere? Does it actually affect the cost? It's a relatively low set current (100uA) se we are somewhat sensitive to capacitor leakage if we actually want to achieve the LDO's rated performance (if we don't need the ultra-low noise then we should just use a cheaper LDO).

gkasprow, 2019-03-16:
I used them due to lower noise, especially microphonic noise caused by the piezo effect that is a nightmare in low noise circuits with big X7R capacitors...
If we use big ceramic ones, one can notice vibrations from fans in output spectra :)

hartytp, 2019-03-16:
ACK. Fair enough. I've never seen issues with microphonics in this kind of application, but maybe the  uTCA fans vibrate more than I'm used to

gkasprow, 2019-03-16:
It depends how deep you dig with spectrum analyzer :D
I probably overestimated the influence of this effect on LDO performance, but it's better to be too careful than to ignore it.

hartytp, 2019-03-16:
okay, sounds good to me.

closed 2019-03-16 by hartytp