sinara-hw/Sayma_RTM #64: check HMC830 connections match dev kit

hartytp, 2019-03-16:
Looks okay to me from a quick skim, but would be good for someone to double check

marmeladapk, 2019-04-02:
@hartytp I already checked this during previous design review round and wrote my comments here: https://github.com/sinara-hw/sinara/issues/481.

My only worry is that we're implementing something blindly without understanding these connections. @gkasprow @hartytp What do you think about adding jumpers to each of these pins so that they can be floated?

hartytp, 2019-04-02:
@marmeladapk ack. If you can fit that many 0402 resistors around the chip without layout issues (e.g. we mustn't make critical signals like the loop filter larger to accommodate them) then jumpers are fine.

It's a bit tough to know what to do, since I've had experience with HMC830 giving pins special functions used in diagnostic modes that are only cryptically documented at best. So, reading the datasheet alone, one can end up with a non-working design. Thus, it's usually best to follow the eval board since that can be tested off-line easily. I'd be more relaxed if the HMC830 worked well on *all* of the current Sayma boards, but AFAICT that's not true.

FWIW, it also puzzles me a little that the data sheet explicitly says that all NC pins were grounded during characterisation of the chip, yet they're not grounded on the eval board. I would bet that they used the eval board for characterisation, so the data sheet claim is likely incorrect...

gkasprow, 2019-04-08:
Let's follow the eval board schematic.

closed 2019-04-08 by gkasprow