sinara-hw/Sayma_RTM #68: RTM FPGA needs to see clock before HMC chips

sbourdeauducq, 2019-03-21:
The cascaded HMC chips don't have deterministic phase. The RTM FPGA uses DDMTD to fix that, and it needs a reference. It can use CDR_CLK_CLEAN when that clock is selected, but when it isn't (which will be the case for initial hardware testing) and the external clock SMA is used instead, it's not working.
I suggest routing an output of IC46 to a GC pin of the RTM FPGA.

sbourdeauducq, 2019-03-21:
Also, the signal from that external clock SMA should not be named EXT_100M_RF_IN, since it's not always 100MHz, e.g. we're using the RTIO clock there in the current v1 DDMTD sync setup.

sbourdeauducq, 2019-03-21:
If must be, replace CDR_CLK_CLEAN2 on R1/R2, but it would be nice to have both if possible.

sbourdeauducq, 2019-03-21:
And those two do not *need* to be GCs if the HMC7043 output at the other input of the DDMTD is a GC.

hartytp, 2019-03-21:
> The cascaded HMC chips don't have deterministic phase. The RTM FPGA uses DDMTD and HMC7043 slips to fix that, and it needs a reference

I thought you were doing all that on the AMC FPGA?

sbourdeauducq, 2019-03-21:
No, it's noisy when SAWG is enabled.

sbourdeauducq, 2019-04-04:
@gkasprow Ping. This is important to get the DAC sync working well.

gkasprow, 2019-04-08:
I connected another ref clock from IC46 to MRCC input of the RTM FPGA.

gkasprow, 2019-04-08:
[image]

gkasprow, 2019-04-08:
[image]

closed 2019-04-08 by gkasprow