sinara-hw/Sayma_RTM #69: N12V0A/P12V0A diodes?

hartytp, 2019-03-26:
@gkasprow I think the answer to this is "no" but to double check....

Do we need diodes on the P12V0/N12V0 lines to ensure that they start up correctly (i.e. to avoid the usual issues where one rail turns on too fast and reverse-biases the other one via OpAmp protection diodes etc)? Currently there are no such diodes on BaseMod, but with all channels populated there could be quite a lot of reverse current.

gkasprow, 2019-03-26:
This issue was observed at the output of the LDOs.
I already implemented them.
[image]

gkasprow, 2019-03-26:
I implemented them on BaseMod. Sayma RTM has increased startup time of the  -6V converter so it will start later than +6V

hartytp, 2019-03-26:
What about the p and n 12v?

gkasprow, 2019-03-26:
we did not observe such issue with these rails. I can increase the startup time of -12V converter.

hartytp, 2019-03-26:
Ok. Ime these kinds of issues are quite dependent on what kind of load is present. Personally I’d stick the diodes there to be safe.

gkasprow, 2019-03-26:
do you mean diodes at the DC/DC converter output? such diodes are already part of the buck converters.

hartytp, 2019-03-26:
yes, sorry, you're right. This is an issue for LDOs, not bucks.

closed 2019-03-26 by hartytp