sinara-hw/Sayma_RTM #87: PI analysis

gkasprow, 2019-04-18:
P2V0 rail under 150% load
[image]
We are well within the LDO limit. But current density in some places is too high.
[image]
LDOs were shorted to simulate the whole PDN.
After optimization, the current density does not exceed 55A/mm2
[image]
and DC drop was lowered twice. Note that most of the DC drop is caused by the inductor DCR
[image]

gkasprow, 2019-04-18:
P4V0 PDN after optimization
via current
[image]

current density
[image]

DC drop is mainly on DCR
[image]

gkasprow, 2019-04-18:
P6V0 rail after optimization
DC drop is dominated by inductors DCR
[image]

The current density is slightly high, but lower than 67A/mm2. It cannot be easily lowered.
[image]

gkasprow, 2019-04-18:
N12V0A PDN. Nothing to optimize
[image]
[image]

gkasprow, 2019-04-18:
P12V0A Nothing to optimize

gkasprow, 2019-04-18:
N6V0: 28mV drop, max I density 22.8A/mm2, I optimized the dropout slightly,

gkasprow, 2019-04-18:
P3V3 looked bad before optimization. After some tweaking it is OK:
drop is dominated by inductors DCR
[image]

dnadlinger, 2019-04-18:
Out of curiosity, is this HyperLynx? I'm sure I've seen this UI somewhere before…

gkasprow, 2019-04-19:
@dnadlinger Yes, it is. Very similar results are produced by SiWave, but it does not have the 3D mode.

gkasprow, 2019-04-26:
12V rail before optimization
[image]
[image]

and after some tweaking
[image]

gkasprow, 2019-04-26:
P1V0. Reduction of the drop from 220mV to 57mV
[image]

gkasprow, 2019-04-26:
P1V2
[image]

gkasprow, 2019-04-26:
P2V5
[image]

closed 2019-04-26 by gkasprow