sinara-hw/Sayma_RTM #88: Power supply - AD9154BCPZ

filipswit, 2019-04-24:
Power supply could be improved by: 

- [ ] placing more vias and moving vias before capacitors, or placing solid, same as on oposite side.

[image]

- [ ] placing track PVDD12A through capacitor C37A
[image]

- [ ] placing via before cap C58A, C55A, C54A
[image]

gkasprow, 2019-04-24:
Good catch!

gkasprow, 2019-04-26:
in case of C58A etc not much can be done. But distances are so short that it does not really matter.

closed 2019-04-26 by gkasprow