sinara-hw/Sayma_RTM #89: Internal planes

filipswit, 2019-04-24:
- [ ] GND and power vias should be directly conected to planes.  Thermal reliefs causes dead copper.
[image]

gkasprow, 2019-04-24:
which AD version do you use? Earlier versions can import rules differently.

marmeladapk, 2019-04-24:
Altium 17 has a single rule, which can define different style of connections to through-pad, smd and vias. AD16 and earlier default to this behaviour seen above.

filipswit, 2019-04-24:
16.1, so that coud be caused by older AD.
What about internal plane 1, in AD16.1 is unconnected.

dhslichter, 2019-04-26:
Not to mention that thermal relief is horrible from an SI perspective...

gkasprow, 2019-04-26:
Under AD19 it looks fine.
[image]

filipswit, 2019-04-29:
Thermal reliefs were coused by AD16.1, Now I use A19 and it's fine.
But, Plane P2 isn't connected.
[image]

I don't know what is the purpose of dividing GND plane into two GNDs. It's on 3 negative layers.
[image]

gkasprow, 2019-05-20:
fixed. It was reminiscent from Xpedition version

closed 2019-05-20 by gkasprow