sinara-hw/Sayma_RTM #93: Differential pairs

filipswit, 2019-04-26:
Based on Saturn:

Class | Plane | Impedance
-- | -- | --
MGT | External | 95
  | Internal | 86
LVDS | External | 96
  | Internal | 93

- [ ] Generally there are missing vias connecting reference planes.

- [ ] Helper_DCXO could be improved by omission Mid-2 layer (blue one)
[image]

- [ ] AMC_GT.GT0TX kind of right angle
[image]

- [ ] GT16TX can be straighten
[image]

- [ ] MEZZ0_LVDS5 aren't well coupled
[image]

- [ ] MEZZ0_LVDS6 right angle
[image]

filipswit, 2019-04-26:
Looks like these signals on botom layer don't have solid reference plane under them.
[image]
[image]

gkasprow, 2019-04-26:
we don't really care about power plane below user IO lines.

closed 2019-04-26 by gkasprow