sinara-hw/Sayma_RTM #100: Layout chech cross talk

hartytp, 2019-05-02:
Apologies if this is stating the obvious but: during the layout review please check that all sensitive analog lines (clocks, dac outputs etc) are well separated from noise sources like smps or digital lines. I’m particularly concerned about the afe dio lines (lvds adc) so let’s do everything we can to screen them from the dac, lo, etc

hartytp, 2019-05-17:
@marmeladapk did you check this when you did your design review? If so, can you close this please?

Same for the other open issues.

marmeladapk, 2019-05-17:
@filipswit

hartytp, 2019-05-17:
@gkasprow @marmeladapk what about all the other open issues? Are these boards ready for manufacture? If so, can the other issues be closed?

marmeladapk, 2019-05-17:
@gkasprow needs to close them once they are done.

hartytp, 2019-05-17:
OKay. But, just to check we're on the same page, the boards aren't ready to send off for manufacture until these are all closed, right? What's the timeline for that? When can we get Sayma to the board house???

gkasprow, 2019-05-20:
I went through this issue and optimised a little the layout increasing separation between ADC LVDS lines. 
DAC and LO lines are far away.

closed 2019-05-20 by gkasprow