sinara-hw/Sayma_RTM #115: AFE - maximum currents on power supply rails

marmeladapk, 2019-10-14:
During our stress tests we're quickly approaching > 100 W power consumption and testmods do not even pull as much current as they're supposed to do:

| Rail | Current testmod | Theoretical value on sheet |
|------|---------------------------|-------------------------------|
| +12V | 200 mA | 400 mA |
|-12V|100 mA | 100 mA |
|+6V | 1.5 A | 3 A |
|-6V | 100 mA | 200 mA |
|+3V3| 1 A | 1.5 A |

Even with these lower values we exceeds MTCA spec (80W). We can change resistor values to pull as much as we need, however I'm not sure if this is a good idea as we won't be able to supply that much in the final environment (at least not until next MTCA spec with 120 and 240 W power consumption arrive).

How much current do we really need on AFEs? Could we just tone down these requirements?

gkasprow, 2019-10-14:
The power supply system was not designed to simultaneously draw max current from all rails.
That's why I applied such loads.
We should add a constraint on max power consumed by the AFE.

jbqubit, 2020-01-17:
@marmeladapk What did you decide for limiting current to AFE?