sinara-hw/Sayma_RTM #116: DAC SYSREF signal format

sbourdeauducq, 2019-10-18:
What signal format should be programmed into the HMC7043 for DAC SYSREF?
This says "CML HP" (is that a reference to the "high power" mode of the HMC7043?):
[image]
This says "LVDS" (are the 50 ohm resistors above supposed to convert CML HP to LVDS? Why not use LVDS output directly from the HMC7043 then?):
[image]

sbourdeauducq, 2019-10-18:
And should this one be configured:
[image]

gkasprow, 2019-10-19:
the outputs with resistors to GND are LVPECL, the ones without are CML. We need internal resistors in the case of CML. 100R should work. I'm not 100% sure but LVDS probably were not strong enough to drive the DAC.

closed 2019-10-21 by sbourdeauducq