sinara-hw/Sayma_RTM #121: photos

jbqubit, 2020-03-31:
Please post photos of the stuffed Sayma_RTM for contract accounting.

> Sayma RTM v2.0: print 10 PCBs, stuff 6

akaminska, 2020-04-07:
1907180PW0009A0103
[image]
[image]

akaminska, 2020-04-07:
1907180PW0008A0104
[image]
[image]

akaminska, 2020-04-07:
1907180PW0008A0107
[image]
[image]

akaminska, 2020-04-07:
1907180PW0008A0105
[image]
[image]

akaminska, 2020-04-07:
1907180PW0008A0106
[image]
[image]

akaminska, 2020-04-07:
1907180PW0008A0108
[image]
[image]

akaminska, 2020-04-07:
Unpopulated PCBs
[image]
[image]

akaminska, 2020-04-10:
Please remember that these panels are "prototypes" as well, we definitely aim for better quality for the final product...
[image]
[image]

jbqubit, 2020-04-10:
Looks good. Thank you!

closed 2020-04-10 by jbqubit

hartytp, 2020-05-22:
Did anyone take a nice photo of Sayma AMC + RTM with panels fitted?