sinara-hw/Sayma_RTM #122: Si5324 fails to ack

sbourdeauducq, 2020-07-10:
Sometimes happens on some boards:
```
[     0.011435s]  INFO(satman): gateware ident 6.7258.3f0cf6e6.beta;satellite
panic at src/libcore/result.rs:945:5: cannot initialize Si5324:
"Si5324 failed to ack write address"
```

jordens, 2020-07-10:
The log above was from me. I assumed that it was from the AMC but it looks like that I2C bus works fine (lots of successful SI5324 traffic before the error). And then it starts passing RTM messages. On the RTM there is one I2C xfer, which looks nice (low voltage levels at least during the address portion) on both sides of the I2C switch, but still errors. Didn't investigate further.

marmeladapk, 2020-07-10:
Fixed with https://github.com/m-labs/artiq/pull/1483 + https://github.com/m-labs/migen/pull/212 . @hartytp @jordens Please recompile your build with those changes and reopen if necessary.

closed 2020-07-10 by marmeladapk